gbfans/afterlife-lightkit #3: [FEATURE] Recreation of 1984 Cyclotron Effect

prodestrian, 2022-06-21:
The most basic requirement for this project to get to a first release is the 1984 Cyclotron Effect, effectively a "Hello World" task but a great way for us to refine our processes before jumping into the heavier stuff.

Based on the [schematic](https://github.com/gbfans/afterlife-lightkit/blob/main/HARDWARE/Schematic.pdf), I assume we'll be running `WS2812` LED strips (Neopixels)?

## Acceptance Criteria
- I should be able to compile and upload the Arduino code to the Microcontroller
- The four LEDs should illuminate in a looped sequence (approx 500ms each) as soon as the sketch starts, one at a time

## Out of Scope
These can be added in subsequent updates once the proof of concept is tested and more of the code boilerplate has been implemented:
- Multiple LEDS per Cyclotron light (eg 4x5LEDs, or a 20-LED strip)
- Customization (interval, brightness, direction, number of LEDs)
- Fadeout effects (incandescent-lamp style)

## Proof of concept sketch
(NOTE: Untested as I don't currently have the hardware available, but in theory this should upload and work with minimal changes)
```ino
#include <Adafruit_NeoPixel.h>

// D1 / GPIO5 = Cyclotron Light Data Out
#define CYCLOTRON_PIN 5

// How long to illuminate each Cyclotron LED
#define CYCLOTRON_DELAY 500

// How many LEDS in the Cyclotron
#define CYCLOTRON_LED_LENGTH 4

// Parameter 1 = number of pixels in strip
// Parameter 2 = Arduino pin number (most are valid)
// Parameter 3 = pixel type flags, add together as needed:
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wired for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wired for RGB bitstream (v1 FLORA pixels, not v2)
Adafruit_NeoPixel cyclotron = Adafruit_NeoPixel(CYCLOTRON_LED_LENGTH, CYCLOTRON_PIN, NEO_GRB + NEO_KHZ800);

uint16_t currentCyclotron = 0;
unsigned long previousCyclotronMillis=0;

uint32_t red = cyclotron.Color(255,0,0);

void setup() {
  cyclotron.setBrightness(255); // set brightness

  cyclotron.begin();
  cyclotron.show(); // Initialize all pixels to 'off'
}

void loop() {
  tickCyclotron();
}

void tickCyclotron() {
  if (
    ((unsigned long)(millis() - previousCyclotronMillis) >= CYCLOTRON_DELAY)
    || previousCyclotronMillis == 0
    ) {
    previousCyclotronMillis = millis();
    cyclotron.clear();
    cyclotron.setPixelColor(currentCyclotron, red);

    currentCyclotron++;

    if (currentCyclotron == CYCLOTRON_LED_LENGTH) {
      currentCyclotron = 0;
    }

    cyclotron.show();
  }
}
```

ajquick, 2022-06-21:
Yes this works.

[image]

I also made the following change:

````
// How long to illuminate each Cyclotron LED
#define CYCLOTRON_DELAY 20

// How many LEDS in the Cyclotron
#define CYCLOTRON_LED_LENGTH 40
````

And now it functions as the Afterlife Cyclotron.

[image]

@cfunseth also has written code that already works for Afterlife and has the ramping speed up effect: https://github.com/cfunseth/AfterlifeCyclotron 

If I recall it also has 1984 mode that places the pixels in the correct spot depending on the number of LEDs.

Currently my test ring has 40 leds, but I believe the final version that will ship with the kit has 32 or 35 LEDs.

ajquick, 2022-06-21:
BTW. I really like the idea of having separate branches for things to test maybe? I typically just commit to master, but I think separate branches for each effect being tested or whatever would be good?

The basic bare setup will be easy, it is once we start adding all the different modes and customizations that things will get hard. So if we can find a way to keep those very portable that would be good.

I'm using just an Arduino and LED ring to test as I don't have an ESP8266 on hand at the moment, but it should be easy to swap between the two devices for the time being.

ajquick, 2022-06-21:
I attempted to get the powercell sequence working, however my knowledge of the Neopixel library fell flat. I was unable to turn the LEDs off in reverse sequence.. even though it *should* have done it.

[image]

```c
#include <Adafruit_NeoPixel.h>

// D1 / GPIO5 = Powercell Light Data Out
#define POWERCELL_PIN 3

// How long to illuminate each Powercell LED
#define POWERCELL_DELAY 20

// How many LEDS in the Powercell
#define POWERCELL_LED_LENGTH 40

// Parameter 1 = number of pixels in strip
// Parameter 2 = Arduino pin number (most are valid)
// Parameter 3 = pixel type flags, add together as needed:
//   NEO_KHZ800  800 KHz bitstream (most NeoPixel products w/WS2812 LEDs)
//   NEO_KHZ400  400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
//   NEO_GRB     Pixels are wiblue for GRB bitstream (most NeoPixel products)
//   NEO_RGB     Pixels are wiblue for RGB bitstream (v1 FLORA pixels, not v2)
Adafruit_NeoPixel powercell = Adafruit_NeoPixel(POWERCELL_LED_LENGTH, POWERCELL_PIN, NEO_GRB + NEO_KHZ800);

uint16_t currentPowercell = 0;
unsigned long previousPowercellMillis=0;

uint32_t blue = powercell.Color(0,0,255);
uint32_t off = powercell.Color(0,0,0);

uint16_t powercell_direction = 1;

void setup() {
  powercell.setBrightness(50); // set brightness

  powercell.begin();
  powercell.show(); // Initialize all pixels to 'off'
}

void loop() {
  tickPowercell();
}

void tickPowercell() {
  if (
    ((unsigned long)(millis() - previousPowercellMillis) >= POWERCELL_DELAY)
    || previousPowercellMillis == 0
    ) {
    previousPowercellMillis = millis();
    //powercell.clear();

    if (powercell_direction > 0) {
      powercell.setPixelColor(currentPowercell, blue);
    } else {
      powercell.setPixelColor(currentPowercell, off);
    }

    currentPowercell = currentPowercell + 1 * powercell_direction;

    if (currentPowercell == POWERCELL_LED_LENGTH) {
      powercell_direction = -1;
    }else if (currentPowercell == 0) {
      powercell.clear();
      powercell_direction = 1;
    }

    powercell.show();
  }
}
```

ajquick, 2022-06-21:
[image]

```c
void tickPowercell() {
  if (
    ((unsigned long)(millis() - previousPowercellMillis) >= POWERCELL_DELAY)
    || previousPowercellMillis == 0
    ) {
    previousPowercellMillis = millis();
    //powercell.clear();

    for (int i = 0 ; i < POWERCELL_LED_LENGTH; i++) {
      if (i <= currentPowercell) {
        powercell.setPixelColor(i, blue);
      } else {
        powercell.setPixelColor(i, off);
      }
    }

    currentPowercell = currentPowercell + 1 * powercell_direction;

    if (currentPowercell == POWERCELL_LED_LENGTH) {
      powercell_direction = -1;
    }else if (currentPowercell == 0) {
      //powercell.clear();
      powercell_direction = 1;
    }

    powercell.show();
  }
}
```

prodestrian, 2022-06-21:
Excellent! So, to be completely clear the above code I posted was just to verify that it would work on the hardware, which is why it's so hastily written.
So the next step is to refactor this into something cleaner and more permanent as a starting point to build upon, and that can be our first actual release of the code before we move onto the next feature.
I'm happy to include the POWERCELL feature as part of that, I've already got most of that code written cleanly.

I think the next feature is wrapping our heads around the Configuration, I was going to create a new Issue to track this because it may need some discussion and investigation. I see you've already got something to look at in #4 so I'll have a play when I pick up an ESP8266 later this week.

For that feature, we'd want to be able to manipulate the speed (CYCLOTRON_DELAY) using the web browser.
If we get that minimum requirement working, everything after this point should be smooth sailing because it's all just variations of the same thing.

prodestrian, 2022-06-21:
And yes, separate branches are a good idea, we almost never want anything being committed to `main` directly. I usually lock that branch to force every change to go through a PR (and a code review). Should hopefully reduce the chance of bad code or regressions being introduced.

prodestrian, 2022-06-24:
I finally have an ESP8266 as well now, so tomorrow I'll solder the headers on and get it working with my Neopixels.

I've played with WLED quite a bit, it's why I purchased an ESP32 originally. I don't think we really need it here, but some of the animation code could be handy. 

I'm not worried about running multiple animations at once, I've already got two separate Neopixel strips animating on my pack while _also_ mixing 3 channels of Audio playback and reading/writing packets over Serial. I can be firing and overheating while changing my volume (from the wand) and playing music at the same time. Zero lag if the code is written efficiently.

Here's a video demo of where my animations are at (I'm building a new pack so these will be rewritten in the near future):
https://www.youtube.com/watch?v=F4Oospy0Dfc
There's a fade-up/down on the PowerCell and Cyclotron during startup.
The Cyclotron LEDs also have a fake "incandescent" fadeout when they switch off.

I'm doing all of this with TimedAction: https://playground.arduino.cc/Code/TimedAction/

What I'd like to do is abstract this all out to smaller components so our main sketch only has to be concerned with the "business logic".

Here's a mockup of what we could end up with:
```ino
// 60-LED Neopixel ring, for example
#define CYCLOTRON_LENGTH 60
// Max brightness
#define CYCLOTRON_BRIGHTNESS 255

Cyclotron cyclotron = Cyclotron(CYCLOTRON_LENGTH, CYCLOTRON_BRIGHTNESS);

void setUp() {
    // Initial Speed
    cyclotron.setSpeed(500);
    // Afterlife Mode
    cyclotron.setMode(MODE_AFTERLIFE);
    // Cyclotron is off by default
    cyclotron.disable();
}

void loop() {
    if (packIsActive()) { // Some magic function which reads the state of the power switch
        cyclotron.enable();
        if (packIsFiring()) { // Fire button is pressed
            // Tells the Cyclotron to spin faster
            cyclotron.setSpeed(250);
        } else {
            // Return cyclotron to regular speed
            cyclotron.setSpeed(500);
        }
    } else {
        // Switch off the Cyclotron LEDs
        cyclotron.disable();
    }

    // This is the line which makes it actually work
    cyclotron.update();
}
```

Obviously the above code is completely artificial, but note how we're not really having to deal with the LED animations or timers here, and there's no blocking code either. The Cyclotron class would initiate the TimedAction instance and the Neopixel strip, we don't have to think about it in the main sketch.

We can tell the Cyclotron and PowerCell instances how long we've been firing for, and have them calculate/adjust their own speed accordingly (or maybe keep it simple and just say `cyclotron.setIsOverheating(true)`). There's a lot of possibilities, we just need to imagine them! :smiley: