gbfans/afterlife-lightkit #6: ESP8266 Interrupt Pins

ajquick, 2022-06-22:
An important detail to discuss, know about. The ESP8266 has several interrupts which can be used on nearly every GPIO pin with the exception of D0/GPIO16, When I was designing the schematic, I only had 5 pins left to assign my inputs for control over the lights. 1 of them was output only and 1 of them had no interrupt (D0/GPIO16).

I determined to assign the input signals in the following way:

D0 - Fire input signal
D5 - Video game change mode signal
D6 - Activate venting signal
D7 - Activate lights / enable signal

D5 - D7 are able to have interrupts attached, D0 (fire signal) is not able to.

I decided that it was least important to capture the fire signal immediately. The reason for this is that I imagine when someone presses the fire button, they will be holding that button down so it can be captured at the start of the loop as long as it is still held down. Similarly the activate / enable and venting signal are expected to be on for a fairly long time when they are activated, but are more important than the fire signal.

Lastly the video game change signal is expected to always be a very short burst or short tap. This is actually the only one where an interrupt is very important as we don't want to miss the signal. The rest don't **need** the interrupt, but enable and vent do have the ability to have them if they need.

prodestrian, 2022-06-24:
Do we actually _need_ interrupts though? I've never had any issues with responsiveness, my fire button responds to even the shortest/lightest touch.

I'm not using the ESP8266 in my projects so I could be missing something, but I have a Teensy 3.2 in my want and I read the button state, if the fire button is pressed it sends a string over Serial to the Teensy 3.2 in the Pack, which triggers the lights/sounds. All of this occurs in a tiny fraction of a second, and I've never pressed the button and had it not fire.

As long as we're not introducing delays into our `loop()` I wouldn't have expected us to have any issues here?
I haven't tested the WebServer with a button yet so I don't know if this introduces lag, but apparently the ESPAsyncWebServer (alternative library) is a more optimised alternative if we need it:
https://github.com/me-no-dev/ESPAsyncWebServer#why-should-you-care

Personally I use the `Bounce2` library to handle buttons/inputs and debouncing, so far everything has been incredibly reliable:
https://github.com/thomasfredericks/Bounce2

Again, there could be other factors I'm not aware of, but I still feel like we can get by without these. Everything I've read suggests that it's not a good idea to use Interrupts for buttons (or human inputs), it's usually for things like sensor data.

I'll try to get some headers soldered to my board tomorrow so I can connect some pushbuttons and verify this, I can try pressing buttons repeatedly while making a lot of simultaneous webServer requests and see if any of the presses are missed or delayed. Maybe then connect up my Neopixels and make sure that it's all 100% asynchronous and non-blocking.

ajquick, 2022-06-24:
>Do we actually need interrupts though?

The only thing I can say is *I don't know*. But I did put thought into it that if we need interrupts, those are the pins to put them on. Change mode may be the only input that may need one, but if everything is kept really fast than it might be okay to not need it.

[ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer#why-should-you-care) seems like a no-brainer if it is comparable in memory usage. For 99.9999999% no one will be using the webserver for anything beyond initial setup. Perhaps the web server should shut down if no one connects to it within the first 2-3 minutes? And a configuration setting to enable / disable the web server would also be good.

Personally how I would set this up on my own pack is to wire everything up. Use the Wi-Fi to configure all my options and then turn off Wi-Fi permanently. If I need it back on, then I could connect over USB to re-enable it.

That is unless having 'party mode' where you can have cool light shows come on based upon a click on your phone is highly desirable for end users.

prodestrian, 2022-06-25:
If we wanted we could have a secret button press to switch Wifi on/off. Maybe while the pack is in shutdown mode, hold the Vent button for 5 seconds to activate Wifi, hold for 5 seconds again to shut it down? Slowly blink the Orange Hat light to show when it's active?

We can also automatically enable Wifi if the Configuration (in either EEPROM or LittleFS) is empty. So you can do your initial configuration, but then it skips enabling Wifi afterwards unless you manually activate it.