gbfans/afterlife-lightkit #7: [FEATURE] Recreation of 2021 Afterlife Spinning Cyclotron Effect

prodestrian, 2022-06-24:
Just so we have a place to keep all our discussions/ideas for this feature together once the code for #3 is completed.

The requirement here is that we should be able to replicate the spinning Cyclotron from Ghostbusters Afterlife.

NOTE: I don't recall if any PowerCell animation changes would be needed in Afterlife mode?

## Requirements / Ideas
1. It should look like a spinning lamp (which may mean lighting more than 1 LED at a time)
2. It could spin slower on startup and gradually get faster (like a motor ramping up)
3. The direction should be configurable
4. The speed should be configurable
5. The number of LEDS should be configurable (not everyone will use a 60-LED Neopixel strip)
6. The number of LEDs lit at once should be configurable (the "width" of the light)
7. It could spin faster while firing/overheating
8. It could slow down during shutdown before coming to a stop and switching off
9. The starting position should be configurable (an "Offset", ie "LED 26 is the first LED")
10. It shouldn't interfere with 1984 mode (#3)
11. Each LED could quickly fade out instead of instantly switching off (to emulate a "motion blur" effect)

ajquick, 2022-06-24:
>NOTE: I don't recall if any PowerCell animation changes would be needed in Afterlife mode?

There is one split second where the OGB's packs are showing a special animation in the Powercell while firing. Potentially during overheating. It would certainly be a goal to have, but maybe not a must have for the first release as it was seen so briefly.

prodestrian, 2022-06-25:
> There is one split second where the OGB's packs are showing a special animation in the Powercell while firing. Potentially during overheating. It would certainly be a goal to have, but maybe not a must have for the first release as it was seen so briefly.

Hmm, I can't seem to find that shot...do you know the timecode?

There's a good shot of Phoebe's pack at `01:46:01` (just before Egon appears), the PowerCell appears to pause on the top LED before looping again (it's more "sticky" than the 1984 animation). Seems to match the first shot at `00:48:16` during the proton pack test scene. So we can replicate that easily enough.
[image]

But as far as the OGB packs, I can see a shot at `01:44:54` where what looks like a stunt pack has a special animation on the PowerCell. I notice the Cyclotron lights are all lit up (which isn't possible with the spinning mechanism), and I don't recognise this from any of the other light kits, so it's probably an Arduino based chaser light. We could include this too somewhere (maybe it's a post-firing after-overheating animation, ie while cooling down?)
[image]

prodestrian, 2022-06-25:
I've now got most of my prototyping testbench together, I cut out a cyclotron and power cell from foam core (it's easier than having a full pack on my bench).

Just to see if everything is working I threw together a simple Afterlife animation, currently running it off an Arduino Uno.
It starts slow and gradually speeds up.

I found that adding a diffuser (a sheet of thick paper here) made all the difference in "motion blurring" the LEDs (which sit about 1.5cm away from the lenses).

Video demo: https://www.youtube.com/watch?v=bHUmGDYBWVs

## NOTES
* I'm still using a 1984-style Power Cell animation for the moment.
* I'm lighting 3 Cyclotron LEDs at once because 1 or 2 at a time wasn't bright enough. This is effectively the "width" of the animation.
* These are two separate Neopixel strips running at once (separate pins)

Update: I just uploaded this to my ESP8266 and it's still working great.

ajquick, 2022-06-25:
> But as far as the OGB packs, I can see a shot at 01:44:54 where what looks like a stunt pack has a special animation on the PowerCell. I notice the Cyclotron lights are all lit up (which isn't [...]

That's the one and yes I believe the same thing that it was an overheating type feature. I believe Ben Eadie was asked why it happened and he described it as a mix up with the code being on that pack when it wasn't supposed to be.

prodestrian, 2023-06-12:
I believe we have completed this. Yes, there are some requirements I listed further up but they should be covered by other features (such as being able to change direction, which affects more than just Afterlife mode). For now I don't think it's worth worrying about fading out the LEDs or anything like that, what we currently have looks like motion blur anyway. There's some super niche ideas about the power cell but I think we can come back to those later by adding some additional animations for the accuracy obsessed.

So I'll close this as done for now, we can open new issues later to cover those other ideas if we feel the need.

closed 2023-06-12 by prodestrian