gbfans/afterlife-lightkit #8: Hardware revision, size constraints, connectors?

ajquick, 2022-06-24:
Looking at the components, I realize there is a possibility to actually shrink down the board to a very small size by stacking the boards on top of one another. The current size is around 70.5mm x 54.5mm, or 3842 mm<sup>2</sup>. I believe it is possible to shrink the board down to around 60mm x 30mm or 1800 mm<sup>2</sup> a saving of 65%. Further shrinkage could occur if the board were to have onboard power regulation and the ESP8266 installed directly to the PCB, but I'm not at that stage yet.

Is there any reason to attempt a smaller design? It would not yield any significant price savings.

[image]

### Connectors

While selecting components I decided to use a JST PH connector on the board to connect with the RGB light boards. The reason for this was because the standard cable that is used with the RGB light strips, a JST SM connector, does not come in a PCB mount version! (WHY!?)

So I had gone with the JST PH connector as it seemingly fits the bill and is already in use on the other GBFans light kits.

This has left me negotiating with RGB ring light suppliers and having them custom make the boards with the JST PH connector instead of the SM connector so that they can plug directly into the board.

**But.**

This means that any time additional rings or different sizes or different packages were to be used, they would need to have a PH connector added to the wires so that they can interface to the board.

What if instead of having the JST PH connectors on the board, a short JST SM pigtail harness was soldered to the board directly?

This would save from having to custom make custom rings and instead nearly any off the shelf RGB light using the same connector could be used.

[image]

If this change is done with the SM pigtail, it would also make sense to go through with other changes to make the PCB smaller. The approximately 50 boards I have on order with the V1 design could have the pigtails wired into place without issue as well.

ajquick, 2022-06-24:
Another (albeit temporary) solution would be to use an adapter cable for the first 50 boards.

[image]

Similar to this from Sparkfun, though in 3 pin configuration: https://www.sparkfun.com/products/retired/14708

ajquick, 2022-06-25:
Current design puts the board at 35mm x 50mm (1750mm<sup>2</sup>), with the Wi-Fi antenna extended over the edge of the board for a total length of ~57mm.

[image]

The IDC connector and potentiometer has been moved to the back of the board. I am not 100% sure if we will need the potentiometer if speeds and brightness will be configurable via app. If the potentiometer does anything, I think brightness as a default would be good with the ability to override via Wi-Fi.

It is designed to use the JST SM connectors with strain relief.

Some form of 3D printed case or standoffs can be supplied.

prodestrian, 2022-06-26:
Might be a silly question, but will this board also have wand lights or is it strictly pack lights? I've never used any of the GBFans electronics kits so I'm not sure if we're planning to be compatible with the wand kit? Is that what the IDC connector is for?

I guess the wand kit just sends trigger values for:
* Is Enabled (Active)
* Is Firing
* Is Venting
* Is Mode Changing

Which is fine, but it does mean my idea of using the Orange Hat light to indicate Wifi status won't be feasible (we can use the pack lights, it just means you won't be able to see it if you're already wearing the pack).

It also means the Wand won't communicate "Is Overheating", so we'll need to know the overheat duration that long the wand kit is configured for and make sure our code _exactly_ matches (is it `12 seconds` as per the PDF?). Looks like the wand cools down instantly as soon as you stop firing though, so that's easy.

Just want to make sure I lock all this down before I start getting deep into the code.

ajquick, 2022-06-26:
I will need to double check exactly how the signals are passed through from the wand to the sound kit to the light kit. It may be possible to for example, hold down the fire button and flip the enable switch to activate Wi-Fi but I need to double check that.

You are correct in your understanding for the trigger states, but there is zero ability to send signals back from the light kit. An exception to this may be to use the USB port or serial or some other protocol to communicate, but we don't want to rely upon it..

Here is V2 of the light kit. Still kind of tinkering with it. I added the ability to solder wires directly to the PCB for people who want direct access to the pins on the ESP8266, should there be any reason for that. Serial, analog and reset are not broken out.

[image]

[image]

Since the original intention for this whole project was *just* to do the cyclotron, I originally called this the Afterlife light kit. Because it will be able to do everything for the pack lights including the powercell, I am just calling it the Pack Light Controller now. Thoughts?

Wireless Pack Light Controller or GBFans.com Pack Light Controller or Open Source Pack Light Controller?

prodestrian, 2022-06-26:
> You are correct in your understanding for the trigger states, but there is zero ability to send signals back from the light kit. An exception to this may be to use the USB port or serial or some other protocol to communicate, but we don't want to rely upon it..

I use Serial for my pack, I have two Teensy boards communicating over 2 wires (Tx, Rx). Works great for me, I've never had an issue (latency, dropped packets, etc). But it does add a pretty big layer of complexity to the code, so it's not worth doing unless we plan to get a lot more out of it.

Funnily enough I was just wondering about getting access to the pins. If I had the ability to reassign the LED pins in the code (`D1` and `D2` specifically), it would allow me to use them for `I2C`. Which opens up the possibility of all kinds of fun optional addons.

For example:
**Voltage Sensor**
https://www.adafruit.com/product/904
I've got one of these installed, connected, and calibrated in my pack but I ran out of time to actually do anything fun with it.
The plan was to be able to show my Battery state on the PowerCell LEDs (and/or wand bargraph). If I know the output voltage when the battery is full vs almost empty, I can read this and calculate a battery percentage, and figure out when it's time to recharge my pack.

**LED Matrix Driver**
https://www.adafruit.com/product/1427
I currently use this on my bargraph, but it's also great for running 7-Segment LED displays (I'm using them for an in-progress build of the Delorean Time Circuits Display). Combined with the Cyclotron Neopixels and a modified version of the Afterlife animation, you could definitely use this board as the foundation for an Answer The Call Proton Pack.

> Because it will be able to do everything for the pack lights including the powercell, I am just calling it the Pack Light Controller now. Thoughts? Wireless Pack Light Controller or GBFans.com Pack Light Controller or Open Source Pack Light Controller?

:sweat_smile: Tough decision!
I think the GBFans name is definitely worth retaining (prominently) so it relates back to your existing products/ecosystem. GBFans.com OpenPack Light Kit?

ajquick, 2022-06-26:
> I use Serial for my pack, I have two Teensy boards communicating over 2 wires (Tx, Rx). Works great for me, I've never had an issue (latency, dropped packets, etc). But it does add a pretty big layer of complexity to the code, so it's not worth doing unless we plan to get a lot more out of it.

I'll break out the TX / RX pins too just in case, but they will be less easily accessible. In the future I imagine expanding this with SD card for sound files and a wand kit. That setup could use serial for communication. Maybe use an ESP32 as a USB host in the pack and plug the wand board in with a Type C cable. 

> Funnily enough I was just wondering about getting access to the pins. If I had the ability to reassign the LED pins in the code (`D1` and `D2` specifically), it would allow me to use them for `I2C`. Which opens up the possibility of all kinds of fun optional addons.

I'm not sold on the current pin assignments. I've learned from reading documentation on FastLED and Wled that there are 2 hardware accelerated pins that are perfect for driving these light strips. I think I'm only using one. The other pins will work of course but in bit banging mode. Just a bit slower. I'm not sure if that will affect us at all because we're looking at ~50-60 LEDs max here. People using Wled and others are looking to drive 1,000+ LEDs at 15-30 FPS. 

>GBFans.com OpenPack Light Kit

I like it. No one has used OpenPack before? Perhaps we can have a partnership with the Q-Pack to make one large project.

ajquick, 2022-06-26:
[image]

ajquick, 2022-06-28:
[image]

The prototype boards are in! While I don't have the code to replicate any specific effects, I was able to test functionality and control 3 independent LED light rings at the same time.

This board is expected to be sold with a cyclotron ring only. Customers will be able to add additional lights, or update the firmware to unlock additional features at a later date (once this code is done!). For the purposes of this project the "Afterlife Light Kit" will only be expected to replicate the Afterlife Cyclotron effect.

ajquick, 2022-06-29:
Currently having issues with the enable signal from the GBFans Sound Board. For whatever reason the signal is not able to control the on and off request. I may need to add some form of external pull down resistor as the ESP8266 does not have internal pull downs.

ajquick, 2022-07-01:
I ran into some issues with receiving signals from the GBFans Sound Board. As the Sound Board uses 5V for communication and the ESP8266 uses 3.3V, there must be level conversion. The Powercell board has a 100k pull up resistor and the beta (previous) version of the Afterlife light kit has a bidirectional logic level shifter with it's own pull up resistor. The Sound Board appears to have a 10K resistor which limits sourcing current to just 0.5mA.

When all is said and done the bidirectional logic level shifter overrides the Sound Board's signal. This means that as currently designed the board will not work. Since these signals aren't high speed, I believe using a MOSFET or other unidirectional logic level shifter would be best.

I have designed a simple fix that I hope will work for the "Afterlife Light Kit" version of hardware which will have a very limited run. After the hardware tests are confirmed as working, I will make the changes to the "OpenPack Light Controller" before building a prototype of that design. 

Testing with these logic converters: 
- 74LVC1G07 (single)
- 74VHCT126 (quad)

ajquick, 2022-07-12:
Powercell boards have arrived.

[image]

prodestrian, 2022-07-23:
@ajquick This all looks super cool! And thanks for the boards, it should really help me to be more confident that the code I push actually works with the hardware. Haven't been able to do much these last few weekends but I have some capacity soon to get back into this and hopefully start closing off some tasks.

"OpenPack" looks perfect too :ok_hand:

ajquick, 2022-07-23:
No problem Chris! I'm currently away for a week myself. Hopefully we can get something going toward the end of the month / August. I think I can start shipping kits to customers in August.

ajquick, 2022-07-28:
Because of the pancake design, I figured it would be best to include a 3D printed case with the kit. That way people can throw it in with double sided tape or Velcro and not run the risk of things shorting out. It also helps with the wire management. The wires will have labels indicating Powercell, Cyclotron and Vent as well for ease of use.

[image]

(Shown in PLA, the case will be black PETG).

prodestrian, 2022-08-11:
@ajquick boards just arrived, thanks so much! I'm done with events for now (not suiting up again until October) so I should be able to get stuck back into prototyping this weekend. Really looking forward to it, especially once we get some of these tedious tickets closed off (FX Library and Config Manager especially).

I think the connectors you went with were absolutely the right choice, they look solid :+1:

ajquick, 2022-08-12:
That would be great! I'll try to get a bit more of my code uploaded soon. I started working on the config system and the mode / state switching... but have not yet compiled or tested it.

prodestrian, 2022-08-18:
Just a quick update... I only had a few hours free on the weekend, but I did manage to get the boards up and running. Unfortunately it took me an hour to find a USB cable which was thin enough to fit, the placement of the USB port means it interferes with the LED connector wires. I can't tell if this is something you've adjusted in the latest revision though.

Once I found the cable I connected it and it started running your demo sketch (power cell and Cyclotron looping), no issues there. Lights are a bit dimmer than I expected but I don't know if your demo sketch just has the brightness turned way down, we'll see when I upload new code.

I did notice that the vent LED ring unfortunately won't fit inside a 3D printed Q-pack MK3 N-Filter, I can get exact measurements but the walls are too thick. Won't be an issue for anyone using aluminium parts I assume. If it was close I'd just suggest sanding the edges of the LED ring but it's off by a fair bit. This might be something we need to either document or offer alternatives for.

Anyway, at this stage my upcoming weekend is clear so I can finally upload my existing sketch and see if it works, if it does then I'll continue working on that.

I'm also going to pick up a 16 pin connector tomorrow so I can try to interface with an Arduino, to effectively emulate the gbfans soundboard and send the same signals for each state (as documented elsewhere here). I'll wire in the wand switches/buttons so I can replicate all the same features. 

@ajquick If you're able to push your configuration code I can try integrating that and testing/building on it, but it's not critical for the FX code at this stage.

Can't wait for the weekend!

prodestrian, 2022-08-20:
## Hardware Testing Update - 20th August 2022
It took a few attempts but I was able to upload the latest sketch to the device (I think I was having some USB issues, was fine after I rebooted my laptop). I modified it to ignore the buttons for now and begin in START mode.
The Cyclotron and Powercell pixels are running as expected, so the hardware is definitely all working.
Also I bumped the brightness up to `255` just so I could compare to the LEDS I was using previously. I'm happy with the brightness, the Powercell looks amazing. Cyclotron looks dimmer because we're currently only illuminating a single pixel at a time, when I connect it to the Powercell connector it looks fine. So that's something we'll take care of with the FX updates.

I also connected the Vent LED ring and added it to the code, it's animating successfully.

I'm now assembling a 10-pin IDC cable so I can prototype further but I think you've already confirmed this part of the hardware is working in #15 . So I'll use an Arduino Nano with 3 pushbuttons (Fire, Vent, Change) and 1 switch (Enable) connected and get it sending the same signals that the GBFans sound board sends.