gbfans/afterlife-lightkit #12: Events and Animation Classes

prodestrian, 2022-07-03:
The next feature after #11 is to implement code to handle the Events and Animations.

I propose that we have a set of predefined animations with generic names.
We then have separate classes for the PowerCell and Cyclotron (treating them almost like standalone devices).
We can assign a Mode to each device (Classic, Afterlife, etc).
There would be a list of States, eg "Startup, Shutdown, Idle, Firing, Overheating, Venting".
We will (eventually) use Configuration to map an Animation to a State with a Mode on a Device, ie "Play the Cycling animation backwards at 300ms delay in Red on the Cyclotron when in the Classic Mode while in the Idle state".
The parameters (delay, direction, color) belong outside the animation.

Animations can be generic across both the PowerCell and Cyclotron as we may want the same effect on both devices (with different parameters).
The Classic cylotron animation could be "CYCLING", the Afterlife animation could be "SPINNING".
We need to come up with clear names for the other remaining animations.

Additionally some animations (eg Afterlife SPINNING) will require further parameters, such as width/radius.

We will also need the ability to change speeds on the fly for each animation, including speed up/down. Ideally this could be a method inside the animation class, where we can define a start speed, end speed, and duration (ie "start at 300ms, then speed up to 100ms over 10 seconds").

Brightness control is also important, both as a high-level setting and for individual pixels. For example the Classic startup animation may need to fade up all LEDS, then fade back down again.

We will also need to know when an animation has completed, so that we know when the startup/vent animations have finished and we can move onto the next one.

ajquick, 2022-07-03:
> We will also need the ability to change speeds on the fly for each animation, including speed up/down. Ideally this could be a method inside the animation class, where we can define a start speed, [...]

I believe we should 100% go with the approach @cfunseth did with his [Afterlife Code](https://github.com/cfunseth/AfterlifeCyclotron). 

He used the [RAMP library](https://github.com/siteswapjuggler/RAMP) which functions amazingly. I have been testing and modifying Cole's Afterlife library and have tested the RAMP library extensively now. The way it preforms elastic changes in speed and brightness is amazing.

For example if I flip the switch on, the Cyclotron lights fade in and then the speed ramps up to the idle speed. If I flip the switch off, the Cyclotron starts to slow down and then fade off. But if I decide half way to flip the switch back on, it simply starts to increase in brightness and speed again. It makes the Cyclotron behave very much like a real moving object would. It also has the option for non-linear *ramping* action. So it starts out slow and then accelerates up to the top speed and vice versa. Really, really cool effect.

If you would like I can try to apply the RAMP library to the existing code and see what happens. We would need to add a ```setBrightness``` and ```setSpeed``` function and then update them on each light update. Actually fairly easy I am thinking.

prodestrian, 2022-07-03:
Ah yes I was meaning to look into that, it sounded exactly like what we needed. And it looks like we can keep all interactions with RAMP within the FX classes, which keeps that layer completely hidden from the main sketch. I hadn't realised it also supported easing functions, that opens up a whole lot of additional possibilities.

Absolutely on the same page about the Afterlife cyclotron operating more like a motor spinning up/down, it adds so much more realism!

I've got a few more hours of weekend left so I'll start writing some code now and see how far I get.

prodestrian, 2022-07-03:
PR with proof of concept of this working is here: #14