gbfans/afterlife-lightkit #23: [BUG] WiFi Access Point does not work on most devices

prodestrian, 2023-06-26:
As discussed in #10 , most devices are not able to connect to the Soft Access Point (SoftAP).

* :heavy_check_mark: My Ubuntu Laptop
* :x: My Samsung phone
* :x: My Google Pixel phone
* :x: My MacOS Desktop

These devices see the access point (SSID) but then fail to connect. My mobile devices immediately disconnect with errors (eg "Association Rejection"). My MacOS desktop behaves as if an invalid password was provided.

I have removed all custom code and tried 6 or 7 sample WiFi access point sketches from various vendors.
All behave the same way.

The Webserver demo we had previously was working on my ESP8266 device but as far as I can tell has never worked on the GBFans board (Lolin Wemos D1 Mini Clone?).

I have tried multiple combinations of Arduino IDE settings for the board, including frequencies and compiler options.
I have tried using the Android Studio debugger (via logcat), nothing obvious showed up in the logs.
I have tried various flags, delays, WiFi options, disconnecting before connecting, clearing the flash memory, etc.
I have performed multiple hours of debugging, too many steps to document here.
I have read through 50+ similar-sounding support threads with no success.
I have tried downgrading to older versions of the libraries, again without success.

So this leads me to believe that one of the following must be the case:
* I have faulty hardware
* I have misunderstood what model the board actually is (and have misconfigured platformIO and my Arduino IDE)
* The board we are using doesn't reliably support Access Point (AP) mode
* There is some kind of undocumented workaround which our specific board needs to get this working

prodestrian, 2023-06-26:
## Testing Update

I have the original ESP8266 board I purchased a year ago before my GBFans board arrived.
I uploaded the exact same webserver sketch to it with zero changes.

It showed up in my Android WiFi list. I connected to it and was immediately shown the Captive Portal screen:
[image]

It 100% works on this original board. So I'm very confident there's nothing wrong with the code I've written here.

ajquick, 2024-07-20:
Testing:

[image]

### #⁠1
D1 Mini Clone from Amazon "MakerFocus"
ESP-12 

Desktop: Can connect to AP and access webpage. 
Samsung Phone: Can connect to AP and access webpage.

Output power no changes. Seems to have enough power either way.

### #⁠2
D1 Mini Clone from AliExpress
ESP8622

This is probably the one I sent. It is attached to the board with the full chips. Lets assume this has too high of a power draw to work via USB alone.

Desktop: Can't connect. Network Security Key isn't correct.
Samsung Phone: Can't connect.

Changed TX power to 17.

Desktop: Can't connect.
Samsung Phone: Can't connect.

Changed TX power to 13.

Desktop: Can't connect.
Samsung Phone: Can't connect.

Changed TX power to 10.

Desktop: Can't connect.
Samsung Phone: Can't connect.

Changed TX power to 5.

Desktop: Can't connect.
Samsung Phone: Can't connect.

### #⁠3
D1 Mini Clone (assuming from Amazon)
ESP8622

This one was one I was originally using to test with. The driver for the CH304 was acting funny so I had to downgrade my driver to get it to work again.

Desktop: Can connect to AP and access webpage. 
Samsung Phone: Can connect to AP and access webpage. 

Changed TX power to 10.

Desktop: Can connect to AP and access webpage. 
Samsung Phone: Can connect to AP and access webpage. 

Output power no changes. Seems to have enough power either way.

### My Guess

The board when powered by USB and connected to the full setup (including other chips) is likely not enough to power everything including the power hungry wifi. It is possible this would be fixed when powered correctly from a 12V source or by adjusting the TX power down. What will need to end up happening either way is to swap to a different ESP8266 module. I should be able to do some more testing soon.

ajquick, 2024-07-22:
Tested two more of the #⁠2 boards from AliExpress.

One mounted on a removable header, along with a complete OpenPack Light Controller backpack.

On it's own, it could not connect to Wi-Fi.

Adjusted TX power down to 17. It could connect and toggle.
Adjusted TX power down to 10. It could connect and toggle.

Installed same board on to the OpenPack Light controller backpack. 

It connected briefly at TX power of 17, but it wasn't a really stable connection. Had trouble connecting at any other TX level.

Tested another one of these boards that was hard soldered to the OpenPack board. It could not connect at any of the tested TX power levels.

Lets just say these knock off D1 Mini boards are junk. I am awaiting my 4.0 USB-C versions to arrive soon.

prodestrian, 2024-07-23:
I desoldered the included board and tried it standalone, same WiFi issues. So I agree, they're probably junk. I did have an attempt to review the Android debug logs via Logcat to see if any specific errors jumped out at me but there wasn't anything I could understand there to explain why it would always immediately disconnect from ESPAP and reconnect to my home WiFi.

I plugged an alternative D1 mini into the OpenPack board with the WiFi demo sketch and it works fine. So I don't think there's any reason why the OpenPack board itself might cause issues if it has the right ESP board connected to it.

[image]

I've had a quick look at the differences between the D1 V3 and V4 boards and there's some helpful threads like this:
https://www.reddit.com/r/esp8266/comments/yxsmk1/wemos_d1_mini_v4_first_look_differences/

Most of it sounds good, but a few small things jumped out at me. Firstly several people said most of the V4 boards they received had faulty WiFi, so we need to ensure we find a reputable source, and definitely test WiFi AP on each board (maybe we need a standalone hardware test sketch?). This was a year ago, maybe things have improved. But that's a worry, don't want batches arriving with 2/3 of the boards faulty.

Second thing was related to the USB power not having a diode or fuse any more on the 5V line. If someone had their GBfans kit powered from a Talentcell, and also connected the D1 to USB to upload new code, would this backfeed into their laptop? I know that on my 3D printer I had to use a special **data-only** USB cable to prevent the 5V of my Ender 3's USB port from damaging my Raspberry Pi', hopefully we don't end up in a similar situation. I haven't read all the replies so I don't know if this is something we have to worry about.

If there's any issues with the V4 I'm sure we can find some reasonably priced good quality V3's. The two I have seem to work perfectly despite their low cost.

ajquick, 2024-07-23:
The 4.0 USB-C works from what I can tell. I have not attached one to a OpenPack board yet though.

I'll take a look at the schematic to see about other issues.

ajquick, 2024-07-23:
I agree the lack of a diode on the VBus is a huge liability. This could be fixed on the OpenPack board by installing two P-channel FETs back to back to cut power. Or alternatively, possibly just having the OpenPack board also supplying the 3.3V volts and not connecting to the +VBus power.